MarginallyClever/GcodeCNCDemo #6: Error in scrip when making the verification, GcodeCNCDemo2axis

mboulonne, 2015-11-18:
Please let me know why is this error, thanks in advance.

Arduino:1.6.6 (Windows 7), Carte : "Arduino/Genuino Mega or Mega 2560, ATmega2560 (Mega 2560)"

C:\Users\Michael\Documents\Arduino\GcodeCNCDemo-master\GcodeCNCDemo2Axis\GcodeCNCDemo2Axis.ino: In function 'float atan3(float, float)':

GcodeCNCDemo2Axis:119: error: 'float atan3(float, float)' was declared 'extern' and later 'static' [-fpermissive]

 static float atan3(float dy,float dx) {

                                     ^

GcodeCNCDemo2Axis:41: error: previous declaration of 'float atan3(float, float)' [-fpermissive]

 

       ^

C:\Users\Michael\Documents\Arduino\GcodeCNCDemo-master\GcodeCNCDemo2Axis\GcodeCNCDemo2Axis.ino: In function 'void arc(float, float, float, float, float)':

GcodeCNCDemo2Axis:132: error: 'void arc(float, float, float, float, float)' was declared 'extern' and later 'static' [-fpermissive]

 static void arc(float cx,float cy,float x,float y,float dir) {

                                                            ^

GcodeCNCDemo2Axis:42: error: previous declaration of 'void arc(float, float, float, float, float)' [-fpermissive]

 

      ^

C:\Users\Michael\Documents\Arduino\GcodeCNCDemo-master\GcodeCNCDemo2Axis\GcodeCNCDemo2Axis.ino: In function 'void where()':

C:\Users\Michael\Documents\Arduino\GcodeCNCDemo-master\GcodeCNCDemo2Axis\GcodeCNCDemo2Axis.ino:206:16: warning: deprecated conversion from string constant to 'char*' [-Wwrite-strings]

   output("X",px);

                ^

C:\Users\Michael\Documents\Arduino\GcodeCNCDemo-master\GcodeCNCDemo2Axis\GcodeCNCDemo2Axis.ino:207:16: warning: deprecated conversion from string constant to 'char*' [-Wwrite-strings]

   output("Y",py);

                ^

C:\Users\Michael\Documents\Arduino\GcodeCNCDemo-master\GcodeCNCDemo2Axis\GcodeCNCDemo2Axis.ino:208:16: warning: deprecated conversion from string constant to 'char*' [-Wwrite-strings]

   output("F",fr);

                ^

exit status 1
'float atan3(float, float)' was declared 'extern' and later 'static' [-fpermissive]

  Ce rapport contiendrait plus d'informations si l'option
  "Montrer les informations de sortie pendant la compilation"
  était activée dans Fichier > Préférences.

i-make-robots, 2015-11-19:
This is a problem with the latest arduino ide, not the code. A workaround is to remove the word "static" at the start of the line. Arduino has been notified and has fixed the bug in their code, it should be better by the next release.

mboulonne, 2015-11-20:
Thank for the info, Will try that in a few days....

Best regards

Michaël Boulonne 💡
Chef Eclairagiste
mboulonne@gmail.com
M 450.467.4208
C 514.214.2255

malipetek, 2016-02-02:
Hello, I got the same error, I deleted the static keyword and it worked but only for declaring x y positions not on arcs. So I posted a comment on marginallyclever.com and found this page after googling error message.

i-make-robots, 2016-02-02:
updated.

closed 2016-02-02 by i-make-robots