MarginallyClever/GcodeCNCDemo #7: how to code for X-Y motion if this kind of motion is COREXY or H-BOT

brianbui93, 2015-12-01:
Now, I'm troubled with coding the X-Y motion when my project focuses on the kind of Corexy motion please help me to some instruction about it !. Thank you !

i-make-robots, 2015-12-01:
The makelangelo project already does what you want.

brianbui93, 2015-12-04:
thanks a lot

i-make-robots, 2015-12-04:
http://github.com/marginallyclever/makelangelo-firmware
change #define POLARGRAPH2 to #define COREXY and you should be good to go.
refer to our forums for tech support.

Glad we could help!

brianbui93, 2015-12-12:
There is a problem that I still don't understand in your code. It is the
Bresenham' algorithm to drawing straight line. I have read documents on
Internet but still troubled with them: how to pulse to step motors.
Would you mind if you can give me more detailed about this case ?

P/s: I'm so sorry because I'm a beginner.

i-make-robots, 2015-12-12:
Tell me which part you understand.

brianbui93, 2015-12-12:
I refer to file "GcodeCNCDemo6AxisRumba.ino". I understand almost all of
your code except function which uses bresenham's line algorithm to move
both motors. Can you describe in detail how to apply this algorithm to
control step motor?. For example. I suppose that you only control 2 motors
X and Y simultaneously in which n1, n2 are the steps needed to operate
motor X, Y, respectively, t is the time to move both motors.To simplify, if
ok, can you summarize the algorithm in flow chart ?. Thank you very much !.

misan, 2016-09-29:
Ever head of Google? https://www.google.es/search?q=bresenham+flowchart&tbm=isch&tbo=u&source=univ&sa=X&ved=0ahUKEwiVoYGrrrTPAhWC7RQKHS2lBkgQsAQIHQ&biw=1190&bih=682&dpr=1.25

i-make-robots, 2016-09-29:
Makelangelo-firmware supports hoot and corexy.

i-make-robots, 2016-10-26:
@brianbui93 do you understand bresenham's algorithm now?  I'd like to close this issue.

brianbui93, 2016-11-05:
Yes, I understand this. Thanks Dan Royer and everybody so much :)