MarginallyClever/GcodeCNCDemo #10: logical position

kaiser1609, 2016-06-24:
Hi,
I am using your GcodeCNCDemo to control two stepper motors to move a solenoid into a specific position to move a chess figure.
I worked a lot of things out like implementing a servo to lift up the solenoid and different lights for different players. 
The only problem I need to solve, is the homing at the start of the program.
As far as I know it should be in line 108, but I can't work out how it should look like. 
I don't understand the Marlin Firmware so I can't even try to work out how they fixed the problem.
Could you give me a hint how to work it out?
Thanks already

i-make-robots, 2016-06-24:
G28 gcode is the home command.
Your machine would then move until it touches limit switches and knows its position. (Eg it measures where it is)

Another option is to move the machine to a known location and use G92 Xa Yb
Where "a" is the known X value and "b" is the known y value.

Does that help?

kaiser1609, 2016-06-25:
Thanks a lot! 
I bought a new switch and a 100k resistor today and now it works...

I will try the G28 straight away :-)

i-make-robots, 2016-06-25:
Pics!  Viiiiiideoooooo!

i-make-robots, 2016-10-26:
@kaiser1609 pics?  video?

closed 2016-10-26 by i-make-robots