MarginallyClever/GcodeCNCDemo #11: infinite loop in parsenumber()

i-make-robots, 2016-08-31:
float parsenumber(char code,float val) {
  char *ptr=buffer;
  while(ptr && *ptr && ptr<buffer+sofar) {
    if(*ptr==code) {
      return atof(ptr+1);
    }
    ptr=strchr(ptr,' ')+1;
  }
  return val;
} 

if strchr() does not find ' ' it returns null (0) and ptr becomes 1.  while(ptr) would not stop the loop.

while(ptr... should be while(ptr>1...

i-make-robots, 2016-08-31:
possible fix in 42c0b5d65d51f3352612709255cd09155de4eb32

misan, 2016-09-29:
just ptr++; worked for me.

Trouble is strchr returns null if space is not found.

i-make-robots, 2016-09-29:
See makelangelo-firmware for updated version.

i-make-robots, 2016-10-26:
updated all versions to match makelangelo-firmware, but now github desktop is broken in windows 10.

i-make-robots, 2016-10-26:
updated all branches to the makelangelo-firmware version.  please confirm it plays nice with the compiler.

commit 50a325f1 (2017-07-28): updated parsenumber() to match makelangelo-firmware

should fix #11
closed 2017-07-28 by i-make-robots