MarginallyClever/GcodeCNCDemo #13: GcodeCNCDemo4AxisRAMPS 5axis

darenwai, 2016-10-26:
Hi. Thank you for your code. I tried GcodeCNCDemo4AxisRAMPS.ino  with no problem.
Since ramps1.4 have two extruder. I wanted to use the E0 and E1 as well so that we can have 5axis.

I changed the whole sketch to take E1 as G letter but nothing happened to E1. Motor not moving.
I know the pin number of E1 and I know how to add the 5axis argument as I am a programmer.

I wonder if there is a limitation in the sketch that are unable to active the 5axis?

bercho, 2016-10-26:
It has no limitations for axis but double extruder requires firmware that recognizes T0 and T1 code for switching between extruder.

darenwai, 2016-10-26:
Hi. Can you take a look at my code. All the 4 axis moving correctly. Only the 5axis didn't move. Dunno what have I missed. The 5axis is G. So you can type G00 X5.0000 Y5.0000 Z5.0000 E5.0000 G 5.000 F50.000
Code: https://drive.google.com/file/d/0BzSicdTSN7f9UnRvMXIwaHVlWFk/view
If you make it work. Then it be updated to GitHub to fully utilize ramps1.4

darenwai, 2016-10-26:
i am changing the G to H for the 5axis. If it work. I will let you know later.

darenwai, 2016-10-26:
It works :)

closed 2016-10-26 by darenwai

i-make-robots, 2016-10-27:
Nice work, @darnwai & @berran remzi