MarginallyClever/GcodeCNCDemo #20: Help required on modifying GcodeCNCDemo4AxisRAMPS.ino to support 5 axis

balajisah, 2016-12-14:
Hi!

I downloaded the GcodeCNCDemo4AxisRAMPS.ino from your github site and modified to add 5th stepper motor.

When i try to run, i here humming noise from stepper motors.
I tried all combinations of steps and feed rate.

Enclosed is my modified file.

Please help me with values of feedrate that will work.

Also help me with the logic behind the feed rate calculation and step delay.

What is the feed rate in g00 command.

Thanks & Regards

Balaji

balajisah, 2016-12-14:
[GcodeCNCDemo4AxisRAMPS.zip](https://github.com/MarginallyClever/GcodeCNCDemo/files/652263/GcodeCNCDemo4AxisRAMPS.zip)

i-make-robots, 2016-12-14:
There was already a six axis version.

balajisah, 2016-12-14:
Will the 6 axis version version work with RAMPS1.4

drf5n, 2022-04-01:
> Will the 6 axis version version work with RAMPS1.4

It will is you set the pins properly in 

https://github.com/MarginallyClever/GcodeCNCDemo/blob/master/GcodeCNCDemo6AxisRumba/GcodeCNCDemo6AxisRumba.ino#L398-L427

i-make-robots, 2022-04-02:
Ramos has room for 6 drivers? Xyze…?

drf5n, 2022-04-02:
A Ramps 1.4 board has sockets for drivers for 5 axes, X,Y,Z, E0, E1, but exposes lots of pins you could hook a 6th axis  driver onto.  The SERVOS header, AUX-2 and AUX-4 are wide open.  You could jumper 9 pins from AUX-4, plus 5V and GND to the 12 pin header on a CNC Shield V3 and socket drivers for 9 axes.

[image]