MarginallyClever/GcodeCNCDemo #22: iso c++ forbids comparison error

ekaggrat, 2017-07-26:
i get this error on compiling gcodedemo2axis

iso C++ forbids comparison between a point and a integer

i-make-robots, 2017-07-26:
in file what on line number what..?

ekaggrat, 2017-07-26:
haults here

 while(ptr>1 && *ptr && ptr<buffer+sofar) {

in this function

float parsenumber(char code,float val) {
  char *ptr=buffer;
  while(ptr>1 && *ptr && ptr<buffer+sofar) {
    if(*ptr==code) {
      return atof(ptr+1);
    }
    ptr=strchr(ptr,' ');
  }
  return val;
} 

but i guess the porblem is somewhere else

the same function compiles okay in the 4 axis version!

i-make-robots, 2017-07-26:
the 4 axis version is probably fixed and has a subtle difference.

ekaggrat, 2017-07-28:
i got it working but i am not able to jog the motors. I works ok on a gcode file but just wont jog with a G0 command. M18 M17 M114 work fine .. what iam i doing wrong?

i-make-robots, 2017-07-28:
all gcodecncdemos have been updated with the new parseNumber used in makelangeloFirmware.  Please let me know if that helps.

commit 4087a503 (2017-07-28): updated all parseNumber() to fix #22
closed 2017-07-28 by i-make-robots