MarginallyClever/GcodeCNCDemo #23: 'serialBuffer' and 'parsenumber' are not declared in this scope

bjrnschmdt, 2018-01-05:
Hello,
I get the following messages when trying to compile the code for the 2-Axis Demo:

GcodeCNCDemo2Axis.ino: In function 'float parseNumber(char, float)':
GcodeCNCDemo2Axis:180: error: 'serialBuffer' was not declared in this scope
GcodeCNCDemo2Axis.ino: In function 'void processCommand()':
GcodeCNCDemo2Axis:239: error: 'parsenumber' was not declared in this scope

What can I do to fix this?

Thank you very much for taking the time to write the code and to make it available for everyone interested.

All the Best, Björn

bjrnschmdt, 2018-01-05:
Okay, i think i know now.

You took the clean code from Makelangelo (as reported in issue 22 i think) where have 'parseNumber' but in the demo there ist only 'parsenumber'. Almost the same with 'serialBuffer' from Makelangelo, but there is only 'buffer' in the demo. I now changed the code to work with 'Makelangelo' standard.

Sorry, for opening the thread/issue as some should not see a difficulty in this error message. But for those like me who are just start experimenting with Arduino and coding i thought it might be helpful.

Regards, Björn

i-make-robots, 2018-01-05:
Thank you for solving it.  Please send a pull request with your changes.

rcKeith, 2018-09-12:
Hi
I get this error 
serialBuffer' was not declared in this scope
on the "4 Axis CNC Demo  - supports MEGA 2560 Arduino RAMPS 1.4"
Do I need to load another Library.  Trying to compile on Ardiuno 1.8.5

or am I missing something
Thanks
Keith

i-make-robots, 2018-09-12:
You missed nothing.  My bad.  I've pushed an update with fixes.

i-make-robots, 2018-09-12:
@bjrnschmdt I forgot about this thread.  Next time you make changes please send me a pull request so your name appears as a contributor in the commits.

Stay awesome!

closed 2018-09-12 by i-make-robots

rcKeith, 2018-09-12:
Hi Dan
Thank you compiles perfectly now.

Keith