MarginallyClever/GcodeCNCDemo #24: undefined references when trying to compile 2 axis gcode.

AvdHdn, 2018-01-05:
I am trying to get the 2 axis code to work on an arduino UNO. 

Whilst trying to upload/compile the sketch i get the error that a couple of functions are an undefined reference. These are the four references which have "extern void" before them in the conig.h. Error message:
<b>C:\Users\arjan\AppData\Local\Temp\ccA5BIH1.ltrans0.ltrans.o: In function `line(float, float)':

D:\Users\arjan\Documents\Arduino\DR\GcodeCNCDemo-master\GcodeCNCDemo-master\GcodeCNCDemo2Axis/GcodeCNCDemo2Axis.ino:92: undefined reference to `m1step(int)'

D:\Users\arjan\Documents\Arduino\DR\GcodeCNCDemo-master\GcodeCNCDemo-master\GcodeCNCDemo2Axis/GcodeCNCDemo2Axis.ino:96: undefined reference to `m2step(int)'

D:\Users\arjan\Documents\Arduino\DR\GcodeCNCDemo-master\GcodeCNCDemo-master\GcodeCNCDemo2Axis/GcodeCNCDemo2Axis.ino:103: undefined reference to `m2step(int)'

D:\Users\arjan\Documents\Arduino\DR\GcodeCNCDemo-master\GcodeCNCDemo-master\GcodeCNCDemo2Axis/GcodeCNCDemo2Axis.ino:107: undefined reference to `m1step(int)'

C:\Users\arjan\AppData\Local\Temp\ccA5BIH1.ltrans0.ltrans.o: In function `processCommand':

D:\Users\arjan\Documents\Arduino\DR\GcodeCNCDemo-master\GcodeCNCDemo-master\GcodeCNCDemo2Axis/GcodeCNCDemo2Axis.ino:271: undefined reference to `disable()'

collect2.exe: error: ld returned 1 exit status</b>

I have attached a .zip file with the demo code and the config.h I am currently using. The only thing I have changed to the code is the fix for the serialBuffer and parse(N)number, which has been described in another thread. 

I hope somebody can help me get this code working. 
[GcodeCNCDemo2Axis.zip](https://github.com/MarginallyClever/GcodeCNCDemo/files/1608138/GcodeCNCDemo2Axis.zip)

i-make-robots, 2018-01-05:
looks like it didn't read config.h, or it can't find the other files in the project.

AvdHdn, 2018-01-09:
Hey!
Thanks for the help!
I looked for an issue as you described and you are right. 
I plan on only using the "Gcode reading" part of the code, not the control of the steppers. Hence, I deleted that part. In the deleted part were the external functions the program couldn't find in my issue. 

I reattached the files to the code and all was well. Thank you!

closed 2018-01-09 by AvdHdn