MarginallyClever/GcodeCNCDemo #27: Is it possible to use with TB6560 stepper driver modules

rcKeith, 2018-08-14:
Hi
I've been Hot wire CNC'ing  for a several years now and get quite a lot of feed back on my website https://www.rckeith.co.uk  Your GcodeCNCDemo for 4 axis on the RAMPS look very good, but could you use the Arduino Mega without the RAMPS board and plug the TB6560 modules in to it.
I use NEMA23 on my machine with a parallel controller.  It look lik a  RAMPS board with the  A4988 or DRV8825 drivers can't handle the current on my steppers, rated at 2.8Amps.
I've dabbled with Arduino, but not that much.  I just retired from work as a Linux System Admin so I'm fairly OK with coding, but perhaps I need to get more into Arduino.
Thanks for nay advice
Keith

i-make-robots, 2018-08-14:
I have used TB6560s before with RAMPS and RUMBA boards.  I use an adapter of my own design, [Sir ramps-a-lot](https://www.tindie.com/products/i-make-robots/sir-rampsalot/).  The latest version sends the step/dir/ena/gnd to the existing pins/screw terminals on the board, which makes it dead simple to connect an external high-current driver to the microcontroller you already know and love.

rcKeith, 2018-08-14:
Hi DanThanks for the very quick reply. Thats excellent news, I'll be ordering some of those. 
Keith

"@type": "EmailMessage",
"@type": "ViewAction",
"target": "https://github.com/MarginallyClever/GcodeCNCDemo/issues/27#issuecomment-412966518",
"url": "https://github.com/MarginallyClever/GcodeCNCDemo/issues/27#issuecomment-412966518",
"name": "View Issue"
},
"description": "View this Issue on GitHub",
"publisher": {
"@type": "Organization",
"name": "GitHub",
"url": "https://github.com"
}
},
{
"@type": "MessageCard",
"@context": "http://schema.org/extensions",
"hideOriginalBody": "false",
"originator": "AF6C5A86-E920-430C-9C59-A73278B5EFEB",
"title": "Re: [MarginallyClever/GcodeCNCDemo] Is it possible to use with TB6560 stepper driver modules (#27)",
"sections": [
{
"text": "",
"activityTitle": "**Dan Royer**",
"activityImage": "https://assets-cdn.github.com/images/email/message_cards/avatar.png",
"activitySubtitle": "@i-make-robots",
"facts": [

]
}
],
{
"name": "Add a comment",
"@type": "ActionCard",
"inputs": [
{
"isMultiLine": true,
"@type": "TextInput",
"id": "IssueComment",
"isRequired": false
}
],
"actions": [
{
"name": "Comment",
"@type": "HttpPOST",
"target": "https://api.github.com",
"body": "{\n\"commandName\": \"IssueComment\",\n\"repositoryFullName\": \"MarginallyClever/GcodeCNCDemo\",\n\"issueId\": 27,\n\"IssueComment\": \"{{IssueComment.value}}\"\n}"
}
]
},
{
"name": "Close issue",
"@type": "HttpPOST",
"target": "https://api.github.com",
"body": "{\n\"commandName\": \"IssueClose\",\n\"repositoryFullName\": \"MarginallyClever/GcodeCNCDemo\",\n\"issueId\": 27\n}"
},
{
"targets": [
{
"os": "default",
"uri": "https://github.com/MarginallyClever/GcodeCNCDemo/issues/27#issuecomment-412966518"
}
],
"@type": "OpenUri",
"name": "View on GitHub"
},
{
"name": "Unsubscribe",
"@type": "HttpPOST",
"target": "https://api.github.com",
"body": "{\n\"commandName\": \"MuteNotification\",\n\"threadId\": 368281764\n}"
}
],
"themeColor": "26292E"
}
]

closed 2022-04-02 by i-make-robots