MarginallyClever/GcodeCNCDemo #28: GcodeCNCDemo, not working

athulsnair, 2019-02-22:
Thanks for sharing how to write a gcode intepretor. I tired GcodeCNCDemo code in my Arduino Mega+RAMPS board using A4988 and DRV8825 stepper motor drivers. I uploaded the code, I get the Info message But then I tried to move motors
This is the command I used 
G00 X10 Y10 F500;

But it doesn't work.

i-make-robots, 2019-02-22:
turn on newline?

athulsnair, 2019-02-25:
Actually the code doesn't even compile on Arduino?? Has some error like Bsome place code uses serialBuffer which isn't even defined. Same error for parseNumber
GcodeCNCDemo4AxisRAMPS works fine

i-make-robots, 2019-02-25:
I haven't looked at this code in a very long time.  The arduino app has
changed to be more restrictive, and now this code probably doesn't get away
with some of the tricks it used.

athulsnair, 2019-02-26:
Can you take a look at the code.

I'm learning to program micro controllers, so I thought first I could compile and make sure code works in Arduino, then change each Arduino function to my own function using register level coding.

i-make-robots, 2019-02-26:
This code is best for reading to compare with the other versions and learn
about the differences.
For production use the better choice is to use Makelangelo-firmware or
something like Marlin or GRNL

athulsnair, 2019-02-28:
I have been using Marlin and GRBL for my 3d printer and CNC. I have always wondered how it works. That's when i came across this code. Like I said I'm still learning, currently understanding GRBL and MARLIN is very difficult for me. I thought I would start small and move to advanced set up and this was a starting point. 4 axis code doesn't have a ARC movement setup.
If possible please take a look at the code

i-make-robots, 2019-02-28:
generally arc is built on top of lines.  later versions of the firmware
(such as makelangelo-firmware) implement g02 and g03 arcs.

athulsnair, 2019-03-01:
Okay. I see the firmware in your git hub page. But to learn everything I would require a makelangelo robot. Is it open source???

i-make-robots, 2019-03-01:
Yes!  So yes.  https://www.thingiverse.com/thing:2494228

athulsnair, 2019-03-04:
GcodeCNCDemo4AxisRAMPS didn't have an arc functionality, but I was able to use the ARC code from GcodeCNCDemo4axisV2 

You already familer with the code, even though here's the code
```
// This method assumes the limits have already been checked.
// This method assumes the start and end radius match.
// This method assumes arcs are not >180 degrees (PI radians)
// cx/cy - center of circle
// x/y - end position
// dir - ARC_CW or ARC_CCW to control direction of arc
static void arc(float cx,float cy,float x,float y,float dir) {
	// get radius
	float dx = px - cx;
	float dy = py - cy;
	float radius=sqrt(dx*dx+dy*dy);

	// find angle of arc (sweep)
	float angle1=atan3(dy,dx);
	float angle2=atan3(y-cy,x-cx);
	float theta=angle2-angle1;
	
	if(dir>0 && theta<0) angle2+=2*PI;
	else if(dir<0 && theta>0) angle1+=2*PI;
	
	theta=angle2-angle1;
	
	// get length of arc
	// float circ=PI*2.0*radius;
	// float len=theta*circ/(PI*2.0);
	// simplifies to
	float len = abs(theta) * radius;

	int i, segments = ceil( len * MM_PER_SEGMENT );
	
	float nx, ny, angle3, scale;

	for(i=0;i<segments;++i) {
		// interpolate around the arc
		scale = ((float)i)/((float)segments);
		
		angle3 = ( theta * scale ) + angle1;
		nx = cx + cos(angle3) * radius;
		ny = cy + sin(angle3) * radius;
		// send it to the planner
		line(nx,ny,pz,pe);
	}
	
	line(x,y,pz,pe);
}
```
I tested it as follows

G00 X0 Y200
G2 X200 Y0 I0 J0

[image]

I moves like arc, but speed is very low. Why is that?

Also where can I find more information on various algorithms used in Marlin & GRBL. In your article you mentioned about Bresenham's line algorithm.
What about look ahead algorithm etc

i-make-robots, 2019-03-04:
Please send a pull request so your change gets added.

I learned about Marlin by reading the code.  it used to be a lot clearer
what they were doing.  now it's very optimized and hard for new programmers
to get into.

athulsnair, 2019-03-05:
I think there's some issue with Arc code. It does what it supposed to do but very very very slowly. Is it supposed to be that way?
if not any idea why?.
You can find the Arc code used in my previous comment.

once I figure that out I will issue a pull request

i-make-robots, 2019-03-05:
Try reducing the number of subdivisions.  it may be overwhelming the system
by having a huge number of extremely short segments.

athulsnair, 2019-04-08:
Reducing #define MM_PER_SEGMENT  (1) improved speed slightly.
I don't know how to issue pull request, i will look in to that

Till then updated code can be found here

[](https://github.com/athulsnair/GcodeCNCDemo)

drf5n, 2022-04-01:
> Actually the code doesn't even compile on Arduino?? Has some errors, some lines code uses serialBuffer which isn't even defined. Same error for parseNumber. I have corrected that errors. When sending move command, motor power turns off. Running M114 returns new position, but no movement of X & Y
> GcodeCNCDemo4AxisRAMPS works fine

These are repaired and tested in https://github.com/MarginallyClever/GcodeCNCDemo/pull/33

closed 2022-04-02 by i-make-robots