MarginallyClever/GcodeCNCDemo #31: Braking problem

MohmmedSwan, 2021-03-07:
Hi
I am using the code [GcodeCNCDemo4AxisCNCShield]
I am unable to control the amount of braking
Can you help me with how to control the amount of braking?

i-make-robots, 2021-03-08:
What do you mean, "braking"?  There are no brakes.

MohmmedSwan, 2021-03-08:
I mean, when Axis stops, it stops hard
Causes large CNC vibration

MohmmedSwan, 2021-03-08:
https://user-images.githubusercontent.com/74132935/110308606-329ead00-7fb5-11eb-8411-b70496ba4d75.mp4

i-make-robots, 2021-03-08:
It’s a demo for education. It has no acceleration or fine tuning. It can’t because the adafruit hardware is not sophisticated enough.

What are you making? I can suggest more appropriate electronics.

MohmmedSwan, 2021-03-08:
I want to make such a pattern

https://www.youtube.com/watch?v=SxPYACoQNnk

i-make-robots, 2021-03-08:
Would you say that's a 3 axis CNC?  If yes then I'd suggest an arduino mega
with a RAMPS shield and you can run many kinds of open source firmware like
GRBL, Marlin, or Makelangelo firmware and they all understand acceleration
- smooth start/stop, higher speed in the middle.

MohmmedSwan, 2021-03-09:
The form consists of 4 axis CNC

closed 2022-04-02 by i-make-robots