MarginallyClever/GcodeCNCDemo #32: Steppers don't turn, just vibrate

MudrakBalazs, 2022-02-05:
Hi!
First off, I just want to thank you for creating this awesome project, I really appreciate it!
But unfortunately I have run into an issue.

I am using the GcodeCNCDemo4AxisCNCShield with Arduino Uno and a CNC Shield v3.0
When I send some GCODE to the Arduino with the Serial Monitor the motors don't turn, they just vibrate/buzz.
And my question is, what could be causing that? Have I not configured something, did I miss a step somewhere? I have confirmed that the wiring and hardware works using GRBL, so that can't be the problem

**Video:**
https://user-images.githubusercontent.com/33597575/152638911-0a5f1f6e-c925-4647-94f8-6572b7c71100.mp4

Thanks in advance for the help! :D

i-make-robots, 2022-02-05:
1. Motor wires crossed, coils fight each other.
2. Stepping speed too high.
3. Low power (unlikely)

What are you making?

MudrakBalazs, 2022-02-06:
Thanks for the reply!
The wiring is good, everything worked with GRBL all the motors worked fine
The motors get sufficient power.
It probably is the stepping speed. How can I adjust that? What variable do I have to change?  (Sorry I'm not a coder, I barely ever touch code)

I'm building an open-source photogrammetry 3D scanner and I think it'll be pretty cool when it's finally finished :D
[image]

i-make-robots, 2022-02-06:
When you send a move command add an F value for feed rate. M114 will tell you the current position and feed rate.

Without a background for reference points it will be very hard to stitch the images. Might be easier to move the camera around the subject and use the image exit data like normal stitching apps.

MudrakBalazs, 2022-02-07:
Hi! Sorry for the late reply I had a busy day. 
I have figured out what's wrong: There is no delay between setting the step_pin HIGH, and setting it low
So originally it's: 

digitalWrite(motors[motor].step_pin,HIGH);
digitalWrite(motors[motor].step_pin,LOW);

but this doesn't work, because there is no delay between the two states.
So the correct way is:

digitalWrite(motors[motor].step_pin,HIGH);
delayMicroseconds(500);
digitalWrite(motors[motor].step_pin,LOW);

You can of course put whatever number you want instead of 500. It is still possible that I'm missing something, but the interpreter is working perfectly.

The turntable will have a non-repeating patter. [(example)](https://i.pinimg.com/originals/f5/0a/ac/f50aac16a5deb91a367827f0b83a137c.png)
We(a mate and I) done previous scans with a white background and Meshroom handled it great.
I'll reach out to you once we have something that's in a mostly working state if you are interested. :D
Current state: 
[image]

closed 2022-02-07 by MudrakBalazs